hansjny/Natural-Nerd #7: [SoundReactive2] Fading does not work properly

hansjny, 2018-11-17:
When switching to fading mode on the sound reactive, sometimes it jumps to other colors

seriousrob, 2018-12-29:
I have looked into your code to rebuild the project. I think i found the solution for your fading issue.

Checkt the function below. It works without jumping between colors for me on an ESP32.

void chillFade() {
  static int Anfang[3] = {0, 234, 255};
  static int Ende[3] = {255, 0, 214};
  static int Schrittweite = 512;
  static double SW[3];
  static double ZW[3];
  static bool reverse = false;
  static int counter = 0;
  if (fade == false) {
    for (int i = 0; i < NUM_LEDS; i++) {
      leds[i] = CRGB(Anfang[0], Anfang[1], Anfang[2]);
    }
    for(int i = 0; i<3; i++){    
      SW[i] = double((Ende[i] - Anfang[i])) / Schrittweite;
      ZW[i] = Anfang[i];
    }
    fade = true;
  }

  if (ZW[0] < Ende[0] && !reverse) 
  {
    for(int i = 0; i<3; i++){    
      ZW[i] += SW[i];
    }
  }
  else
  {
    for(int i = 0; i<3; i++){    
      ZW[i] -= SW[i];
    }
  }

  for (int j = 0; j < NUM_LEDS; j++){
    leds[j] = CRGB((int)round(ZW[0]), (int)round(ZW[1]), (int)round(ZW[2])); 
  }
  FastLED.show(); 
  delay(5);

  counter++;
  if (counter == (int)Schrittweite) {
    reverse = !reverse;
    if(reverse){
      for(int i = 0; i<3; i++){
       ZW[i] = Ende[i]; 
      }
    }else if(!reverse){
      for(int i = 0; i<3; i++){
       ZW[i] = Anfang[i]; 
      }
    }
    counter = 0;
  }
}

magicwoody, 2019-04-12:
runs better now, vy good, tnx

SPrajc, 2019-05-16:
This is perfect, it also removes the ocasional flicker <3

magicwoody, 2019-05-16:
5 Lights working fine
[image]

adryyanbb, 2019-05-26:
This is an amazing project. Unfortunately, I don't think I got the right code

vaibhav4verma, 2019-06-07:
sketch_jun07a:4:29: error: reactive_common.h: No such file or directory

help me please

closed 2019-06-15 by hansjny