hansjny/Natural-Nerd #15: Leds turn on and still white

Danilodo82, 2019-06-29:
hello, I programmed esp32, the blynk application communicates correctly but when I connect the LEDs to esp32, light up and remain white. can you help me please?
P.S. I used a 5v led strip

Netto09, 2019-07-14:
Question 1: I am trying to control by the Blynk application but it did not work. Any idea what changes I should make in the code?
Question 2: How to change chorma in "void chillFade ()"?

Netto09, 2019-07-14:
To use Blynk, I inserted all the codes used by the application libraries and changed the WiFi.mode (WIFI_AP) to "wifi_ap_sta" in master.ino. But it did not work.

P.S .: I am using NodeMCU and Wi-Fi connection for communication with application.

hansjny, 2019-09-29:
Cleaning old issues, closing this.

closed 2019-09-29 by hansjny