hansjny/Natural-Nerd #22: Reactive_common.h no such file or direcrory

QuaserDaser, 2020-04-11:
Hi,
When I try compiling the led_slave or master this error occures. I included all libaries but I dont know what to do with the reactive_common.h. where do I have to put it so it includes it?
Thanks in advance.

CARPER1963, 2020-04-15:
Hola, en la misma carpeta que esta el archivo .iso

closed 2020-04-17 by hansjny

amercer86, 2020-05-02:
> 
> 
> Hi,
> When I try compiling the led_slave or master this error occures. I included all libaries but I dont know what to do with the reactive_common.h. where do I have to put it so it includes it?
> Thanks in advance.

I can't get this to work either. :(

akhil-intronerd, 2021-04-30:
Hey guysIi found the solution for this, its pretty simple actually. There is a file included in the main folder named 'reactive_common.h'
Just open Arduino IDE and goto Toolbar>Sketch>Add File
then locate & select 'reactive_common.h' file from the folder included and done!

akhil-intronerd, 2021-04-30:
Hey guys! I found the solution for this, its pretty simple actually. There is a file included in the main folder named 'reactive_common.h'
Just open Arduino IDE and goto Toolbar>Sketch>Add File
then locate & select 'reactive_common.h' file from the folder included and done!
Also you'll have to do this while uploading both Master & Slave files as it opens this library in other tab in Arduino IDE.

akhil-intronerd, 2021-04-30:
Hey guys! I found the solution for this, its pretty simple actually. There is a file included in the main folder named 'reactive_common.h'
Just open Arduino IDE and go to Toolbar>Sketch>Add File
then locate & select 'reactive_common.h' file from the folder included and done!
Also you'll have to do this while uploading both Master & Slave files as it opens this library in other tab in Arduino IDE.