hansjny/Natural-Nerd #26: Memory Issue for Nano

descriptorvar, 2020-05-23:
Arduino: 1.8.9 (Windows 10), Board: "Arduino Nano, ATmega328P"

In file included from C:\Users\Windows_Admin\Documents\Arduino\Mask_Animation\Mask_Animation.ino:2:0:

C:\Program Files (x86)\Arduino\libraries\FastLED-3.2.9/FastLED.h:14:21: note: #pragma message: FastLED version 3.002.009

 #    pragma message "FastLED version 3.002.009"

                     ^

Sketch uses 14574 bytes (47%) of program storage space. Maximum is 30720 bytes.
Global variables use 2107 bytes (102%) of dynamic memory, leaving -59 bytes for local variables. Maximum is 2048 bytes.
Not enough memory; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing your footprint.
Error compiling for board Arduino Nano.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

PalantirWielder, 2020-05-23:
Have you tried decreasing the LED count (number of rows & LEDs per row) and recompiling? Looks like your Nano doesn't have enough SRAM to handle the current number of LEDs.

descriptorvar, 2020-05-25:
Thank you for your help. For Nano I had to reduce amount of LEDs

closed 2020-05-25 by descriptorvar