hansjny/Natural-Nerd #28: Face mask project

descriptorvar, 2020-06-17:
Is there any schematic how to connect all together. Tried it multiple ways but does not work.

bennyd93, 2020-06-27:
Hi descriptorvar,

I had the same issue, the video really isnt much help. as majority of things are hidden ;/

Attached is a REALLY crude drawing that shows how i managed to get the majority working, although I'm still stuck on how to wire in the push button to flick betwen animations.

hope this helps.
[image]

bennyd93, 2020-06-30:
Hi Descriptorvar (and anyone stuck on this project), 

I have successfully completed the wiring and done a more thorough diagram, so that anyone can see it. In my project I took power and ground for the LEDs off of the 5v and gnd pins as i was only using 256 Leds (8x32 LEDS)
**Note: you will have to change the variables in the ardiuno code and face mask webapp code if you use a different amount to the ones in the video** 

Power for the project comes in via inbuilt usb port

**Momentary Button** (Pins should be used on the same side)
Button pin 1 = Arduino Pin GND (Black wire in diagram)
Button pin 2 = Arduino Pin 2 (Green wire in diagram)

**SD Card Pins** (you must use all connecters)
CS = Arduino Pin SS (Purple wire in diagram)
SCK = Arduino Pin SCK (Yellow wire in diagram)
MOSI = Arduino Pin MOSI (Brown wire in diagram)
MISO = Arduino Pin MISO (Light Blue wire in diagram)
VCC = Arduino Pin 5V (Red wire in diagram)
GND = Arduino Pin GND (Black wire in diagram)

**LED Panel Pins**
5v = Arduino Pin 5v (Red wire in diagram)
GND = Arduino Pin GND (Black wire in diagram)
DATA = Arduino Pin DATA (Green wire in diagram)

### Parts List

Arduino Micro
https://www.amazon.co.uk/gp/product/B019SXN8QM/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1 

SD Card Reader 
https://www.amazon.co.uk/gp/product/B07SBR9NL1/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1

LED Panel (note I used this as to save time soldering and for ease)
https://www.amazon.co.uk/gp/product/B07KT1H481/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

To bring all this together I did use some protoype board so it was more permanant 
https://www.amazon.co.uk/gp/product/B082KYCJXJ/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

Power Bank (Like the video)
Button & wires I had from previous project 

Hope this helps you and anyone else wanting to do this project :)

[image]

descriptorvar, 2020-07-01:
Thank you so much!!!!!!!

descriptorvar, 2020-07-01:
Benny,
I am working with Nano, trying to adopt it to nano

bennyd93, 2020-07-06:
Descriptorvar,

Try this website it tells you what pins to use on a nano, just trace the pins on the diagram to the nano pins :). 

hope this helps

https://www.theengineeringprojects.com/2018/06/introduction-to-arduino-nano.html

HeroFabio, 2020-07-09:
i copy the code but cant verify.
error:
Arduino: 1.8.13 (Windows 10), Board: "Arduino Nano, ATmega328P"

In file included from C:\Users\fabio\AppData\Local\Temp\arduino_modified_sketch_487043\sketch_jul09a.ino:3:0:

C:\Users\fabio\Documents\Arduino\libraries\FastLED-3.3.3/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.003

 #    pragma message "FastLED version 3.003.003"

                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\fabio\AppData\Local\Temp\arduino_modified_sketch_487043\sketch_jul09a.ino: In function 'void setup()':

sketch_jul09a:19:23: error: expected ')' before ';' token

 #define CHIP_SELECT SS;

                       ^

C:\Users\fabio\AppData\Local\Temp\arduino_modified_sketch_487043\sketch_jul09a.ino:142:20: note: in expansion of macro 'CHIP_SELECT'

   while (!SD.begin(CHIP_SELECT))

                    ^~~~~~~~~~~

sketch_jul09a:19:23: error: expected ')' before ';' token

 #define CHIP_SELECT SS;

                       ^

C:\Users\fabio\AppData\Local\Temp\arduino_modified_sketch_487043\sketch_jul09a.ino:142:20: note: in expansion of macro 'CHIP_SELECT'

   while (!SD.begin(CHIP_SELECT))

                    ^~~~~~~~~~~

sketch_jul09a:142:31: error: expected primary-expression before ')' token

   while (!SD.begin(CHIP_SELECT))

                               ^

exit status 1

expected ')' before ';' token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

hansjny, 2020-07-09:
> i copy the code but cant verify.
> error:
> Arduino: 1.8.13 (Windows 10), Board: "Arduino Nano, ATmega328P"
>
> In file included from [...]

Code bug. My bad. Fixed.

closed 2020-07-09 by hansjny

hansjny, 2020-07-09:
Thanks for the writeup @bennyd93 . Added a readme with it, and put a link to it in the source code. I'm sure many will have great use for it.  [Find it here](https://github.com/hansjny/Natural-Nerd/blob/master/arduino/readme.md)