hansjny/Natural-Nerd #35: Connection Dropping

frankjcjr, 2020-08-17:
Hi all,

Hans, this was a great video, and a great summer project so thank you for that. I just got mine up and running today and I'm very excited! However, in sound reactive mode, the connection between master and slave drops every couple minutes, and the light goes back into the single green LED "waiting for connections" mode. It quickly reconnects after a few seconds. Did you/anyone experience this? Not sure if its badly soldered connections to the Wifi boards or if it's something else.

I will add that I am new to coding, and I only attempted this project because the code was already written.

Thanks for your help!

Siranjib, 2021-04-19:
Hi, I am also having the same issue. Have you fixed that already...Can you help me out.

gulliver103, 2021-09-09:
Anyone has solved the issue with the Connection Dropping every now and then?
Thanks for help.

Siranjib, 2021-09-10:
Nah...nobody solved it

RobNO97, 2021-11-01:
Still trying to find a fix... No reply from Nerdforge nor on youtube, nor on his Discord.
The strange thing I noticed is that you can see the event "wifi evt: 7" in the wifi serial debugging which means "WIFI_DISCONNECT_REASON_NOT_ASSOCED".

But this event sometimes also popup without the disconnection issue.

Still didn't quite understand where the problem is